bird-sanctuary/bluejay #6: Layouts for BB51

stylesuxx, 2022-09-01:
Add new layouts for BB51 - see [BLHeli_S Repo](https://github.com/bitdump/BLHeli/commit/59382f3b33a5c1da95405204904eb6d129e9c4f4)

stylesuxx, 2022-09-08:
[26. of May](https://github.com/bitdump/BLHeli/commit/59382f3b33a5c1da95405204904eb6d129e9c4f4):

The commit is a bit polluted since it also includes indentation changes, here the important parts:
* [Booloader start address changed](https://github.com/bitdump/BLHeli/commit/59382f3b33a5c1da95405204904eb6d129e9c4f4#diff-e7e524ba3222f4912ded1354611b568e31465cb83bbadcef759e8d2477777837R6) - Careful, this might require changes in the flash process too.
* [BB51 does not seem to need VDD monitor to be enabled](https://github.com/bitdump/BLHeli/commit/59382f3b33a5c1da95405204904eb6d129e9c4f4#diff-e7e524ba3222f4912ded1354611b568e31465cb83bbadcef759e8d2477777837R60), [2](https://github.com/bitdump/BLHeli/commit/59382f3b33a5c1da95405204904eb6d129e9c4f4#diff-43285af38898d9ac9026a0a2cffa3178fcebd5b20db20abcf1677569a505ce1bR3518)
* [Address check adaptiation according to different bootloader position](https://github.com/bitdump/BLHeli/commit/59382f3b33a5c1da95405204904eb6d129e9c4f4#diff-e7e524ba3222f4912ded1354611b568e31465cb83bbadcef759e8d2477777837R240)
* [Pinout](https://github.com/bitdump/BLHeli/commit/59382f3b33a5c1da95405204904eb6d129e9c4f4#diff-43285af38898d9ac9026a0a2cffa3178fcebd5b20db20abcf1677569a505ce1bR135)
* [Doc](https://github.com/bitdump/BLHeli/commit/59382f3b33a5c1da95405204904eb6d129e9c4f4#diff-43285af38898d9ac9026a0a2cffa3178fcebd5b20db20abcf1677569a505ce1bR167)
* [Include new layouts](https://github.com/bitdump/BLHeli/commit/59382f3b33a5c1da95405204904eb6d129e9c4f4#diff-43285af38898d9ac9026a0a2cffa3178fcebd5b20db20abcf1677569a505ce1bR286)
* [Different EEPROM address](https://github.com/bitdump/BLHeli/commit/59382f3b33a5c1da95405204904eb6d129e9c4f4#diff-43285af38898d9ac9026a0a2cffa3178fcebd5b20db20abcf1677569a505ce1bR517)
* [Different address](https://github.com/bitdump/BLHeli/commit/59382f3b33a5c1da95405204904eb6d129e9c4f4#diff-43285af38898d9ac9026a0a2cffa3178fcebd5b20db20abcf1677569a505ce1bR571)
* Checks for BB21 or BB51: Basically all checks in the form of ```IF MCU_48MHZ == 1``` need to be adjusted to ```IF MCU_48MHZ >= 1```
* [Address offset](https://github.com/bitdump/BLHeli/commit/59382f3b33a5c1da95405204904eb6d129e9c4f4#diff-43285af38898d9ac9026a0a2cffa3178fcebd5b20db20abcf1677569a505ce1bR4581)
* [EFM8 BB51 defines](https://github.com/bitdump/BLHeli/blob/59382f3b33a5c1da95405204904eb6d129e9c4f4/BLHeli_S%20SiLabs/SI_EFM8BB51_Defs.inc)

Layout includes:
* [A_X](https://github.com/bitdump/BLHeli/blob/59382f3b33a5c1da95405204904eb6d129e9c4f4/BLHeli_S%20SiLabs/A_X.inc)
* [B_X](https://github.com/bitdump/BLHeli/blob/59382f3b33a5c1da95405204904eb6d129e9c4f4/BLHeli_S%20SiLabs/B_X.inc)
* [C_X](https://github.com/bitdump/BLHeli/blob/59382f3b33a5c1da95405204904eb6d129e9c4f4/BLHeli_S%20SiLabs/C_X.inc)

Misc:
* [Set_Pwms_Off MACRO changed in the L layout](https://github.com/bitdump/BLHeli/commit/59382f3b33a5c1da95405204904eb6d129e9c4f4#diff-1e339237cd20fe389bf9f483acc0f4bb424573dd33b57496f2d265191dbbbda2R441)

[3. of July](https://github.com/bitdump/BLHeli/commit/6c221ae719d6cb030072c6c790f6ae0b393e85cc):

* [Bugfix address](https://github.com/bitdump/BLHeli/commit/6c221ae719d6cb030072c6c790f6ae0b393e85cc#diff-f153108229a0f84fbd5ce181af1b9df4343d5512f9fd2ce8960f2f5e1968fa86R170)

iggut, 2022-10-02:
Test A_X test board (normal and bi-dir mode) : Completed on a BetaFPV Meteor75 Walksnail edition. It uses their 1S F4 5A v2 AIO flight controller. A_X_5_48_.hex was the file that I used for flight tests.

stylesuxx, 2022-10-02:
@iggut - awesome, thanks for your feedback.

closed 2022-11-10 by stylesuxx