bird-sanctuary/bluejay #9: "Variable" PWM

stylesuxx, 2022-09-01:
Allows the PWM frequency to be adjusted as a setting and does not require the firmware to be re-flashed.

This is actually also part of the branch mentioned in #8

Quick-Flash, 2022-09-14:
As you know the codebase better, is it possible to set a pwm value that is not a 24, 48, or 96khz value?

stylesuxx, 2022-09-15:
Well, 12kHz you could set - everything else, not without major changes.

PCA0PWM is the register you have to set, CLSEL (Cycle Length Select) has three bits but only 4 possible values corresponding to: 8, 9, 10, or 11 bit.

Which other values would you be interested in and why?

Quick-Flash, 2022-09-15:
> Well, 12kHz you could set - everything else, not without major changes.
> 
> PCA0PWM is the register you have to set, CLSEL (Cycle Length Select) has three bits but only 4 possible values corresponding to: 8, 9, 10, or 11 bit.
> 
> Which other values would you be interested in and why?

Well, non integer values would be the first step in proper variable pwm. Personally 12khz could be interesting, I had one toothpick that was unflyable on anything but 24khz, and running that low would be quite interesting to see for some aircraft.

So from what you've said it sounds like some sort of counter is set that determines how long before the next time it fires. Sounds like it could be possible to make it non 12, 24, 48, 96 khz. Maybe one of these days I need to give assembly another try.

stylesuxx, 2022-09-15:
EFM8 does not have a FPU. You can for sure run different duty cycles than that. You should really check out the reference manual, it has the formula on how to calculate settings for different duty cycles. But, you are opening a massive can of worms. Dithering for example is hard coded for each frequency, so making it really dynamic will be a huge quest.

Quick-Flash, 2022-09-16:
I wonder how it would act if we quickly switch pwm freq.

stylesuxx, 2022-09-16:
I don't think this will work at all, from what I see this will affect other functionality too and will most likely mess things up. That's why it's done in the setup phase - once. You need to set one register, then set the PWM register, then again reset the first register and reconfigure quite some stuff. I recommend you take a look at the commit in the PWM branch, this should give you a feeling for what would be needed to change PWM freq on the fly.