bird-sanctuary/bluejay #16: Detach the repo

stylesuxx, 2022-09-11:
Running a fork does seem to come with a couple of downsides, the biggest one being that the fork is not searchable.
Once the stars [if the fork surpass the stars of the original repo, it is possible to search the fork with a special parameter](https://stackoverflow.com/questions/33626326/how-to-search-a-github-fork-for-code-using-the-github-search-api/33666719#33666719).

This kind of sucks badly since it will not be easy to surpass the stars of the original Repo.

The only way to fix this is to detach the repo from the forked one, there seems to be a support form to do so.

@saidinesh5, @damosvil, @Burdalfis any thoughts about this or better ideas? I kind of don't want to lose the reference to the original project, so if we decide to detach, we have to prominently display that we are a fork. Or should I try to contact github support and try to explain the situation?

saidinesh5, 2022-09-12:
Reference to the original project can be put in the Title of the repo for eg. and locally we can always use another git remote to track the original repo.

Plus we sent an invite to matthias to this repo right? So detaching this repo from the original repo shouldn't change much. So if searchability is useful (it usually is for me) let's go ahead and detach the repo.

stylesuxx, 2022-09-12:
> Plus we sent an invite to matthias to this repo right? So detaching this repo from the original repo shouldn't change much. So if searchability is useful (it usually is for me) let's go ahead and detach the repo.

Yup, invited him multiple times. Searchability is also important for the documentation since I link to the search from there in some places to reference code (as to not have to track absolute links)...

Quick-Flash, 2022-09-14:
I agree, with moving forward with this. Github tries to treat forks as though you have them for the purpose of adding to the repo you forked from, or keeping your own private fork of the repo that you just has a few changes and you want to easily keep up to date with the repo you forked from. Its better to just quit being a "fork" and just disclose in the main readme that you are a fork of the original bluejay which is a fork of blheli_s. Just my 2 cents.

stylesuxx, 2022-09-15:
OK, thank you. Your input gives me enough confidence that it is the right thing to do.

stylesuxx, 2022-09-16:
I put in a request to get the repo detached.

commit 99948727 (2022-09-16): chore: Add a section about ancestry. Resolves #16
commit dee1df6f (2022-09-16): chore: Add a section about ancestry. Resolves #16 (#17)
closed 2022-09-17 by stylesuxx