bird-sanctuary/bluejay #21: Motor twitching around 83000rpm

damosvil, 2022-09-22:
### Describe the issue

When motors are spinning at 83.000 rpm they start twitching. It happens when motors have no load, so they can reach those speeds.

```
Bluejay 48khz bidir               ESC1 twitching                                            ESC2 twitching
0802 19500kv                       85000rpm - 1421 throttle - 1,1A                   83000rpm - 1491 throttle - 1,1A
1102   19000kv                       83000rpm - 1474 throttle - 1,5A                 83000rpm - 1544 throttle - 1,2A

Bluejay 48khz no bidir           ESC1 twitching                                          ESC2 twitching
0802 19500kv                        1386 throttle                                                   1456 throttle
1102   19000kv                        1474 throttle                                                   1561 throttle

Bluejay 24khz bidir               ESC1 twitching                                            ESC2 twitching
0802 19500kv                       83000 rpm - 1474 throttle - 1,8A               84000rpm - 1500 throttle - 1,6A
1102   19000kv                       83000 rpm  - 1561 throttle - 2A                 84000rpm - 1600 throttle - 1,8A

Bluejay 24khz no bidir           ESC1 twitching                                          ESC2 twitching
0802 19500kv                        1474 throttle -  1,8A                                       1500 throttle - 1,6A
1102   19000kv                        1561 throttle - 2A                                           1600 throttle - 1,8A

Blheli_S                                      ESC1 twitching                                         ESC2 twitching
0802 19500kv                         1500 throttle - 1,7A                                      1600 throttle - 1,7A
1102   19000kv                         1600 throttle - 2A                                        1600 throttle - 2A
```

Notes: 
    - 19000kv Motors get unstable at 83000rpm, but they get so hot that they can only withstand those rpm for short periods of time (10s and they burn on touch).
    - Tests are all without load
    - Test voltage is 8V
    - Tests made with DSHOT300, Bluejay version PWM 24 & 48khz, Blheli_S PWM 24khz
    - Using Bluejay 48khz with DSHOT600 instead DSHOT300 twitching starts about 80000rpm instead 83000rpm. I would expect a bigger difference, so I think that mcu load is not the real issue for twitching. I think we should take a look with a scope on the comparator output in case there is a physical limitation for commutation. 

Additional tests in other escs have thrown the same issue, so it seems a MCU speed limitation.

### Bluejay version

0.16

### ESC variant

any

### PWM frequency

24

### DShot bitrate

Any

### Bidirectional DShot

Any

### FC firmware

Betaflight  4.3.1

### Motor size

Any

stylesuxx, 2023-06-21:
After juggling some numbers around and speaking to a couple of people (shout out to **TuneRC** in particular) some more findings:

Technically we are limited to around 500k eRPM on 48MHz MCUs, when translating this to our "worst" case that would be motors with 12 poles resulting in a max of 83333RPM (500k * 2 / motorpoles) before the hardware reaches it's limits. If we translate that back to motor kV and 1S HV voltage of 4.35V we arrive at a practical kV limit of 19157kV. This is pretty much our worst case: low pole count, 1S HV.

Since motor KV are usually not given under load, this theoretical limit is to be taken with a grain of salt since the effective RPM will be less under load, thus motor kV always need to be looked at under load - otherwise they are a useless spec standing by themselves and only motors with same dimensions could be compared kV wise.

Nonetheless super high kV motors are pushing the ESC to it's absolute limits. Seeing that there are 32500kV motors it is easy to see, that we are moving into territory where the 83kRPM limits could be reached (in certain edge cases) - thus resulting in de-syncs.

This also explains the twitching observed by @damosvil and is basically by design and not much we can change with the hardware used here.

**Conclusion**: If you insist on running insanely high kV motors, be aware that you are running into real risk of reaching hardware limitations and thus de-syncs. If experiencing de-syncs there are a couple way to mitigate those in this situation:
* throttle limit (this is a bit tricky since flight controller firmware might still send higher throttles values to the  ESC if needed)
* motor output limit (preferred way since this will limit the actual throttle value sent to the ESCs. Reduce this value until de-syncs stop and you should be fine. You can also monitor this by looking at the RPM OSD element and see if you are coming into the critical RPM area)

stylesuxx, 2023-07-19:
Move to Wiki:
* [ ] New page "Limitations"
* [ ] Experiment, explanation, Data, Setup
* [ ] Conclusion about eRPM/RPM limit

stylesuxx, 2023-07-30:
Moved to [Wiki](https://github.com/bird-sanctuary/bluejay/wiki/Known-Limitations) - closed.

closed 2023-07-30 by stylesuxx