bird-sanctuary/bluejay #49: Preparation of build env fails

stylesuxx, 2022-12-26:
```
Run rm -rf ~/.wine
http://: Invalid host name.
Error: Process completed with exit code 1.
```
https://github.com/bird-sanctuary/bluejay/actions/runs/3773393953/jobs/6428016523

saidinesh5, 2022-12-26:
@stylesuxx https://docs.github.com/en/actions/using-workflows/events-that-trigger-workflows#pull_request_target we can change the event from pull_request to this .. since we have to approve actions for random triggers, we should be okay with this..?

stylesuxx, 2022-12-26:
We can try, but I do not think that this will change anything since it is still running on a PR from a fork, which seems to be the main issue here. I will set up a test repo and have a look.

stylesuxx, 2023-01-02:
@saidinesh5 I looked into it, it seems your suggestion is a good solution. And prevents us from leaking secrets. I am going to implement this.

saidinesh5, 2023-01-02:
Yeah.. add me to the issue so i can merge it..

closed 2023-01-03 by stylesuxx