bird-sanctuary/bluejay #53: Temperature control and telemetry wrong on BB21 1S builds

damosvil, 2023-01-02:
### Describe the issue

Temperature telemetry reported on 1S is lower on BB21 1S builds. Seems that the root of the problem might be that on 1S there is no external voltage regulator, so as voltage on 1S is fluctuating ADC has no stable VREF to perform conversions:
[image]

Check different VREF sources and check if at least there is one stable on 1S targets (1.65V might be a starting point).

### Bluejay version

0.18

### ESC variant

O_H_5

### PWM frequency

24

### DShot bitrate

150

### Bidirectional DShot

Any

### FC firmware

Betaflight  4.4

### Motor size

0802

stylesuxx, 2023-01-02:
Related issue in [BLHeli Repo](https://github.com/bitdump/BLHeli/issues/671).

damosvil, 2023-01-03:
@stylesuxx Please, esc-configurator.com tool needs to be updated to add this parameter. Maybe the identifier in the code is not the most appropriate, so do not hesitate to let me know any better one.

stylesuxx, 2023-01-03:
@damosvil yes, created an issue [here](https://github.com/stylesuxx/esc-configurator/issues/312) and gave you some feedback in your PR - maybe we can further discuss in the PR?

closed 2023-02-24 by stylesuxx