bird-sanctuary/bluejay #54: Bug found on Blheli_S calc_next_comm_timing_fast function

damosvil, 2023-01-02:
### Describe the issue

A bug was found in a function imported from Blheli_S: calc_next_comm_timing_fast
https://github.com/bitdump/BLHeli/issues/647

Please, update the code and test it @ 80K RPM

### Bluejay version

0.18

### ESC variant

any

### PWM frequency

Any

### DShot bitrate

Any

### Bidirectional DShot

Any

### FC firmware

Betaflight  4.4

### Motor size

Any

stylesuxx, 2023-01-02:
The issue is a bit of a mess, there is no commit yet that implements the fix, no? Have you looked into it in a bit more detail?

damosvil, 2023-01-02:
At this point this is the only information we have about this bug. It seems to imply a few changes in code, and test it carefully limiting the current of the power supply. I will try it asap on the also known 80k rpm limit to check if there is any improvement or it leads to a regression

stylesuxx, 2023-02-13:
Summoning @garlinplus - you have discovered this bug initially in BLHeli_S, is there any chance that you could have a look into this and see if this still exists in Bluejay. I found the original issue a bit hard to follow...

damosvil, 2023-02-22:
Before fixing this bug I would like to shed light on what this routine does:

```
; Calculate next commutation period
;
; Measure the duration of current commutation period,
; and update Comm_Period4x by averaging a fraction of it.
;
; Note: Comm_Period4x holds the average weighted time of the last 4
; commutations. This routine Removes one the fraction of that time
; and adds the same fraction of the new commutation time. Depending
; on the spinning speed it uses some or other dividers of the formula:
; Comm_Period4x = Comm_Period4x - (Comm_Period4x / (16, 8 or 4)) + (Comm_Period / (4, 2 or 1))
; Above 80k erpm:
; Comm_Period4x = Comm_Period4x - (Comm_Period4x / 16) + (Comm_Period / 4)
; Between 40k to 80k erpm:
; Comm_Period4x = Comm_Period4x - (Comm_Period4x / 8) + (Comm_Period / 2)
; Below 40k erpm:
; Comm_Period4x = Comm_Period4x - (Comm_Period4x / 4) + (Comm_Period / 1)
;
; Simple example using 16 and 4 divisors:
; - Let commutation time be constant.
; - Comm_Period = 64
; - Then Comm_Period4x = 256 (Comm_Period * 4)
; Comm_Period4x = Comm_Period4x - (Comm_Period4x / 16) + Comm_Period / 4
; Comm_Period4x = 256 - (256 / 16) + (64 / 4)
; Comm_Period4x = 256 - 16 + 16
; Comm_Period4x = 256
```

damosvil, 2023-03-26:
Fixed in Bluejay 0.20

closed 2023-04-04 by damosvil