bird-sanctuary/bluejay #78: Turtle mode doesn't work

damosvil, 2023-03-22:
### Describe the issue

On some builds motors don't invert when starting turtle mode

### Bluejay version

0.20

### ESC variant

Any

### PWM frequency

Any

### DShot bitrate

Any

### Bidirectional DShot

On

### FC firmware

Betaflight  4.4, Silverlight

### Motor size

Any

chaotixtech, 2023-03-22:
Happymodel Diamondf4 with icm42688p gyro.
esc type - O_H_5
motor size - 0702 23000kV
pwm freq - dynamic
dshot600
bi-dir - n/a
fc firmware - Quicksilver 0.9.6-dev
No log available

Turtle mode engages correctly but motors fail to reverse and spin at a slower speed than expected with possible fet heating.

damosvil, 2023-03-24:
I was testing on Betaflight 4.5-zulu Mach 17 commit 12530. Tested 0.18.1, 0.19.2, and 0.20 versions in that commit and neither of them work.

On Betaflight 4.4.0 I have tested 0.18.1, and 0.20 and both work.

damosvil, 2023-03-25:
Turtle mode didn't work after rebasing on March 17 4.5-zulu. Try this:
https://github.com/betaflight/betaflight/pull/12276

damosvil, 2023-03-25:
Today I tested with Quicksilver 0.9.5 and Bluejay 0.20 with this Bluejay config, and turtle mode works fine.
Attached you can find my profile configuration for Quicksilver and my Mobula 6 Flysky (Crazybee F4FS lite V4.2) 

I have optimized one of the startup loops that would be the source of the issue. Please find the hex files in the checklist: https://github.com/bird-sanctuary/bluejay/wiki/0.20-Testing-Guide-and-Checklist

[Profile_default_2023-03-25.zip](https://github.com/bird-sanctuary/bluejay/files/11068852/Profile_default_2023-03-25.zip)
[image]

chaotixtech, 2023-03-25:
Tested with Quicksilver 0.9.6-dev and it seems to be working as expected now.

damosvil, 2023-03-25:
Working as spected now in both Betaflight 4.4 and Quicksilver 0.9.5 and 0.9.6-dev

https://discord.com/channels/822952715944460368/1049663643928772700/1089186616968482916

closed 2023-03-26 by damosvil