bird-sanctuary/bluejay #81: Bidirectional operation doesn't work

damosvil, 2023-03-22:
### Describe the issue

When trying bidirectional operation motors doesn't behave as expected

https://discord.com/channels/822952715944460368/1049663643928772700/1088171839278305311

### Bluejay version

0.20

### ESC variant

Any

### PWM frequency

Any

### DShot bitrate

Any

### Bidirectional DShot

Any

### FC firmware

Betaflight  4.4

### Motor size

Any

damosvil, 2023-04-01:
Fixed in:
https://github.com/bird-sanctuary/bluejay/compare/528bb2053167251cfd755bd139fbb48c806b1019..82055a51aaf8cc9b6877cc160dc724864fcbe242
https://github.com/bird-sanctuary/bluejay/compare/82055a51aaf8cc9b6877cc160dc724864fcbe242..b787f4afff2c0a8831852dd53999c24e1dc36168

damosvil, 2023-06-10:
New rework is being done for v0.20

stylesuxx, 2023-07-06:
Confirmed to be working better than ever by the community.

closed 2023-07-14 by stylesuxx