bird-sanctuary/bluejay #83: Weak turtle mode operation

damosvil, 2023-03-26:
### Describe the issue

Some users have reported weak turtle mode operation.

### Bluejay version

0.20 rc1

### ESC variant

Any

### PWM frequency

Any

### DShot bitrate

Any

### Bidirectional DShot

Any

### FC firmware

Betaflight  4.4, Silverlight

### Motor size

Any

damosvil, 2023-03-26:
Looks like that when using dynamic pwm motors don't have enough strength to turn the kwad upside down.

The following patch forces 24khz during user inverted operation (command 21, used in turtle mode). 
https://github.com/bird-sanctuary/bluejay/compare/01b9c6cdcbb877db88113ec837b04624ef645419..b24c462ce5d63daa7c8588a1d1366fb88f2de765

stylesuxx, 2023-03-26:
But this does not affect 3D mode, right? Because Betaflight has settings to compensate for week turtlemode by spinning the props on the opposite side too...

damosvil, 2023-03-26:
> But this does not affect 3D mode, right? Because Betaflight has settings to compensate for week turtlemode by spinning the props on the opposite side too...

I think it doesn't. This bug is related to the blown fets.

damosvil, 2023-03-26:
It worked fine in Tested in the Meteor 65

KondratokGK, 2023-04-26:
Turtle mode not working. It`s wrong motor rotation for turtle mode. It spins as normal mode.

stylesuxx, 2023-04-26:
@KondratokGK please provide more info, this statement helps in no way to debug anything.

Which RC of 0.20.0 are you on exactly? Which flight controller firmware are you using? Which version of FC version? Which ESCs are you using (Layout and dead-time).

damosvil, 2023-05-20:
Scope has been reduced for 0.20

closed 2023-05-20 by damosvil