bird-sanctuary/bluejay #84: Release Notes

stylesuxx, 2023-04-01:
This time so many things happened, I would like to add some more details to the release instead of just the summary of the PRs.

## PWM baked in
PWM is not a setting that can be changed on the fly, no re-flashing of the ESCs is required anymore

## Dynamic PWM
PWM is being changed dynamically during flight. Thresholds can be set by the user via configurator. PWM is being change between 24kHz, 48kHz and 96kHz

## Improved commutation
Commutation has been improved massively, resulting in motros that run much cleaner - you can even hear it. Additionally with EDT this now allows to log the "stress" of a motor in flight.

## Safety features
**EDT safety arm** ensures that the ESC will only arm after a valid EDT handshake. This will only work with flight controller firmware that support EDT functionality.

**RC filtering/smoothing**. As a fallback for bad tunes RC signal filtering has been implemented, this ensures that quick changes in RC signal extremes are smoothed out, prevents motors getting hot and ultimately frying. If you are not using highly experimental tunes the recommendation is to leave filters disabled as it might interfere with what you are expecting from your tune.

## Bugfixes 
* 3D mode is now back to working again after we broke it in 0.17.x

## Quality of life improvements
* Massive code refactoring, everything has been refactored into smaller, more readable modules

rohansx, 2023-04-01:
hey, would love to contribute to this. Assign me.

stylesuxx, 2023-04-01:
@rohansx huh? You want to write the release notes? Sure, feel free to add your comment...

damosvil, 2023-04-02:
In Dynamic PWM section:
PWM is __**not**__ a setting -> PWM is __**now**__ a setting?

closed 2023-07-16 by stylesuxx