bird-sanctuary/bluejay #92: Is that possible to implement BB51 on G-series hardware？

ansmax, 2023-05-19:
HI,team，Is PIN0.0 also can used as ABC/V MUX port? Like G-X-15-24 firmware？
;*********************
; PORT 0 definitions *
;*********************
;			EQU	7
;			EQU	6
RTX_PIN		EQU	5
；     		EQU	4
V_Mux		EQU	3
A_Mux		EQU	2
C_Mux		EQU	1
B_Mux;		EQU	0

stylesuxx, 2023-05-21:
You would need to check the datasheet - it depends on your implementation. Crossbar is a bit more limited on the BB51. IIRC you could have V_MUX on 0.0 B_MUX is not an option on 0.0 though

That being said, you can check the currently available layouts for BB51 - they are well tested and known to be working. 

**Note:** Be aware that we will not add any new layouts for hardware that we did not have in our hands. So should you decide to run your own layout and want it "officially" supported, please join our Discord.

ansmax, 2023-05-22:
> 

Thanks a lot, maybe I should try A/B/C-Series hardware not wasting time on G-Series. May I ask if there is anyway or anywhere to get the schematic circuit diagram that may support BB51? Every  schematic I find only apply to BB21.

stylesuxx, 2023-05-23:
You can re-use the ones for BB21, just map the pins according to the available X layouts and you should be good.

closed 2023-06-05 by stylesuxx