bird-sanctuary/bluejay #94: Losing power gradually until copter lands

stylesuxx, 2023-05-21:
Some users report that they are losing power gradually until they have to land. They can then re-arm and fly again for a limited time. 

I have seen reports of this happening on 0.19. People also report that going back to 0.18 fixed the issue. When asked people said that they did set the power rating to 1S and disabled temp protection.

Hardware this has been happening on:
- Happymodel CrossF4
- Mobeetle 6
- Meteor 75 (At least this one, we can confirm that it's not a general problem - pinging @noctaro - I think you have this one and are not having any issues in that regard?)

Noctaro, 2023-05-21:
Not sure if i changed bj version on the meteor75, will test it tomorrow on 0.19

Noctaro, 2023-05-22:
No problems on my site. Working as expected. But keep in mind its the Meter 75 2022. A-X-5 target. I am not sure if anything changed regarding ESC since the previous Metor 75 version.

stylesuxx, 2023-05-22:
Yeah, thanks for checking. Don't have any more information about the Metor yet. Has not been happening on any of mine eiter, so who knows...

cynfewl, 2023-05-25:
I think I am seeing this issue.

I recently replaced motors on my custom Meteor65 Pro build with a HappyModel SuperbeeF4 (S-H-50) which was running EX0802 19,000kv motors and is now running the new lighter HappyModel 0702 23,000kv motors (black bell). I updated Bluejay 0.19.2 as part of the motor swap and am running Betaflight 4.4.0 with Bidirectional DShot.

For me it's more like a sudden loss of power which I initially thought might be a loose prop or motor bell, but after replacing props and checking the motors it was still happening. Definitely happening more at high throttle and disarming and taking off again the quad would fly ok for at least a couple of seconds before it happened again.

I tried changing motor timing from medium/high to high and increasing min/max startup, but neither fixed it. Power rating was set to 1S and temperature protection set to the default at 140C. I have just flashed back to 0.18.1 and this seems to have resolved the problem. I have min/max startup at the higher values, but all other settings were set back to defaults.

damosvil, 2023-05-25:
@cynfewl could you please try again disabling temperature protection?

cynfewl, 2023-05-31:
Apologies for the delay, but I think temperature protection is indeed the culprit.

I upgraded to 0.19.2 again, but this time disabled temperature protection. Enabling EDT and adding this to the OSD I can see that the ESCs start at about 250 degrees and one ESC (number 3) seems to max out and roll over to 0 or 1. The quad flies fine with the temperature protection disabled and I don't see any loss of power.

I was hoping to post an excerpt from the the DVR, but I can't seem to access the files at the moment. If I manage to access the DVR I will attach some screenshots showing the temps.

damosvil, 2023-05-31:
What is happening is that your aio battery rating is 1-2s so you need to configure esc power rating to 2s+ in esc-configurator.com tool

stylesuxx, 2023-05-31:
Thank you @cynfewl for coming back to us and confirming our suspicions.
Please just confirm that the temp read-outs are correct with the 2S+ power setting. Going forward we will disable temp protection by default and improve the tool-tip in regards to power rating.

cynfewl, 2023-05-31:
This board is actually a Superbee F4 https://www.happymodel.cn/index.php/2022/07/28/superbee-f4-lite-aio-flight-controller-elrs-frsky-version-for-1s-tiny-whoop-drone/ which I believe is 1S only, but I will try it with the 2S setting as well and report back. Apologies, the craft name is a bit misleading.

Regardless I agree a call out in the tool-tip would be handy because I read that section as the cell count you intended to use on the board, not what the board was capable of running, so I would have got this wrong on my CrazybeeX quads that I am running on 1S.

stylesuxx, 2023-05-31:
Ah, this is good to know. I think I have this same flight controller somewhere - I will try to reproduce it.

Yes, the power rating thing is not well explained right now and the truth is, that on a 1S rated board it could very well be 2S+ if they use a step up converter and the MCU's are not directly powered from battery voltage, which is the case most of the time on 1S boards. I think it's a bit difficult to tell users: "Check if your MCU is powered via a regulation circuit", but definitely more context is needed here.

If you could just confirm the behavior on 2S you would save me some time.

cynfewl, 2023-05-31:
I can confirm that setting ESC Power Rating to 2S+ gives me ESC temperatures in the correct range of about 30 degrees. If I run into this issue on other boards I will perform similar tests and add comments here so we can keep track of affected hardware.

I wish I could easily identify how the MCU on these boards is powered, but neither my eyesight nor my electronics knowledge is up to that task I'm afraid.

Thank you very much @damosvil  and @stylesuxx for your help and contributions to this excellent project.

stylesuxx, 2023-05-31:
Well, I think the best place to add this information is to the [Tested Hardware Wiki page](https://github.com/bird-sanctuary/bluejay/wiki/Tested-Hardware).

What's really worth a lot is to know that this is indeed a real issue and we have a solution for it, so thanks again for helping us out here.

**As background info**: Temp protection has never been working properly and has been basically fixed in 0.19. We kept the default to be on, which in hindsight was a mistake, since by it being broken for ever, it technically always was off by default - simply never working and nobody noticed...

stylesuxx, 2023-06-25:
Added explanation on how to fix in the wiki: https://github.com/bird-sanctuary/bluejay/wiki/FAQ#why-is-my-quadcopter-gradually-losing-power-until-it-lands

Future versions of Bluejay will have temperature protection disabled.

closed 2023-06-25 by stylesuxx