bird-sanctuary/bluejay #96: Unify comment types

stylesuxx, 2023-05-24:
In an attempt to unify the code, I suggest the following types of comments and their structure

Types of comments:
* Inline comment - starts at char 35 if possible, prefixed with a single `;`
```asm
    clr ACAB                       ; Comment comes here and relates to this line only
```
* Banner comments in header of a file and to describe a section in great detail should have the following format:
```asm
;**** **** **** **** **** **** **** **** **** **** **** **** ****
;
; HEADLINE
;
; DESCRIPTION 1
; DESCRIPTION 2
;
;**** **** **** **** **** **** **** **** **** **** **** **** ****
```
* Headlines should help  to quickly scan the code and group it into logical sections - only one line:

``` asm
;**** **** **** **** **** **** **** **** **** **** **** **** ****
; HEADLINE
;**** **** **** **** **** **** **** **** **** **** **** **** ****
```

* Label descriptors should be one or multiple lines of comments to describe the label in more detail
```asm
; A short description to provide more context to the label
label:
    ....
```

* Line descriptors should explain what the following line(s) do, it should be aligned with the next line
```asm
play_beep_melody_loop:
    ; Read current location at Eep_Pgm_Beep_Melody to Temp4...
    clr  A
```

damosvil, 2023-06-02:
Agreed

stylesuxx, 2023-06-05:
Added to Wiki as reference: https://github.com/bird-sanctuary/bluejay/wiki/Development

closed 2023-06-05 by stylesuxx