bird-sanctuary/bluejay #127: Dithering re-work

stylesuxx, 2023-06-25:
Dithering is relatively CPU intense and does not seem to have too big of an impact. We should evaluate the current implementation since it does not seem to be optimal.

* Would an optimal solution be worth it?
* How CPU intense would an optimal solution be?
* Would it make sense to rework the current solution?

stylesuxx, 2023-06-28:
I looked a bit into the current dithering implementation:

It seems that dithering is invoked in t1, when the DShot frame is decoded, so dithering is only applied on a per DSHOT frame base, not per PWM cycle. This somehow does not make sense to me. What would make sense to me is to apply dithering every PWM cycle. The current implementation seems to offset the calculated PWM value until the next DSHOT frame arrives.

@damosvil could you by any chance shed some light on this? Am I missing something here?

damosvil, 2023-06-28:
No, you didn't miss anything. This is a substandard dithering implementation. Also this is no 11bit dithering as announced.

stylesuxx, 2024-01-21:
Remove dithering - closing issue.

closed 2024-01-21 by stylesuxx