bird-sanctuary/bluejay #137: Evaluate if bidir braking would make sense as setting

stylesuxx, 2023-07-06:
Might be worth a shot to have this as a setting to improve motor direction reversal in 3D mode.

stylesuxx, 2023-07-08:
This change seems to have brought great improvement on smaller quads: https://github.com/bird-sanctuary/bluejay/pull/136/commits/be25795e6a1e5437b70b86fcb0059505ca498e34 and basically eliminated all de-syncs on motor reversal.

Suggestion: We add this as a setting and **allow the user to set this from 0x10 to 0x50**. Or even better, show the corresponding eRPM.

@damosvil any thoughts on this?

damosvil, 2023-07-08:
For now I would give it more time so people can provide feedback on the current change.
I think that previous setting was tested on 14 pole motors, that are bigger and more powerful than the ones @egonl used in their builds during tests. The new setting should help for less powerful and 12 pole motors, in order to avoid stalls during startup, but I would like to wait until it is tested on bigger motors.
If current settings are generic enough for a wider range of motors I would let it with no parameter, because the common user prefers that things just work out of the box. I would add the setting if we receive feedback from people that are uncomfortable with the current version though.