bird-sanctuary/bluejay #149: Happymodel Mobula 7 elrs - no motor spin on arm on Bluejay 0.20.0 Z-H-30

jawz101, 2023-07-23:
### Describe the issue

https://www.happymodel.cn/index.php/2022/01/20/happymodel-mobula7-1s-micro-fpv-whoop-drone/

upgraded to v0.20.0
motors spin in Betaflight Configurator
ESC detected in esc-configurator
Z-H-30

### Bluejay version

0.20.0

### ESC variant

Z_H_30

### PWM frequency

Any

### DShot bitrate

Any

### Bidirectional DShot

Any

### FC firmware

Betaflight 4.2.2

### Motor size

0802

### Configurator debug log

[esc-configurator-log.txt](https://github.com/bird-sanctuary/bluejay/files/12140171/esc-configurator-log.txt)

stylesuxx, 2023-07-23:
So when the motors spin in BF and you can see them in esc-configurator, what's the issue?

jawz101, 2023-07-24:
When I arm the quad disconnected from my computer, the motors do not spin

stylesuxx, 2023-07-24:
And you have increased min/max startup Power in ESC settings? You might need to drag both sliders completely to the right.

The motors should eventually start spinning if you add throttle. You might need to increase idle throttle too. Please also post a diff from your BF settings.

jawz101, 2023-07-25:
I am such a moron.  I'd been reflashing everything and for some reason I had my receiver set to TAER instead of AETR so my throttle was on the wrong stick.  Set it back to AETR and I'm good.  Sorry about that.

closed 2023-07-25 by jawz101

stylesuxx, 2023-07-25:
Great to hear that you got it going!