bird-sanctuary/bluejay #161: Release Notes v0.20.1

stylesuxx, 2023-08-11:
# v0.20.1

## Bugfixes 
* Bootloader was using wrong keys leading to at state where it would not be possible to save settings or flash firmware. This was only an issue after flashing v0.20.0 via C2 interface. During regular updates via [esc-configurator.com](https://esc-configurator.com) the bootloader is not written and people will not be affected.
* There was a bug where max startup power has been ignored. This is now fixed and will help prevent damage to motors/ESCs in cases where the motors are stuck.

stylesuxx, 2023-08-16:
RC1: [Bluejay_0.20.1-rc1.zip](https://github.com/bird-sanctuary/bluejay/files/12359650/Bluejay_0.20.1-rc1.zip)

damosvil, 2023-08-30:
Please, also add that when bringing back bidireccional 3d mode startup motor protection was unadvertedly disabled in 0.20

Prajwal-Koirala, 2023-12-27:
**Here's a summary of the key points from the release notes for Bluejay firmware version 0.20.1:**

**Bug Fixes:**

- **Bootloader Issue:** Resolved a critical bug that prevented saving settings or flashing firmware after flashing v0.20.0 via the C2 interface. Users who updated via esc-configurator.com are not affected.
- **Startup Motor Protection:** Reinstated startup motor protection, which was inadvertently disabled in v0.20.0, leading to potential hardware damage.
- **Max Startup Power Bug:** Fixed a bug that ignored max startup power settings, potentially causing motor/ESC damage in cases of stuck motors or during crash recovery. This bug was present in both BlheliS and Bluejay up to v0.20.0.

**Important Recommendations:**

- **Update to v0.20.1:** It's strongly recommended to update to v0.20.1 to address these critical bugs and ensure optimal protection of your hardware.
- **Set up Startup Power:** Carefully configure min/max startup power using the provided setup guide to minimize the risk of hardware damage. This is crucial for protecting your motors and ESCs.

**Additional Information:**

- **Affected Users:** Those who flashed v0.20.0 via the C2 interface should prioritize updating to v0.20.1.
- **ESC Configurator Users:** Users who updated via esc-configurator.com are not affected by the bootloader issue.
- **Crash Recovery and Stuck Motors:** The fixed max startup power bug is particularly relevant in crash scenarios or when motors become stuck, as it can help prevent excessive power application and subsequent hardware damage.

stylesuxx, 2024-01-07:
@damosvil @hyp0dermik-code I updated the Changelog - this is now a bit of a mashup of the planned 0.20.0 and 0.21.0 version. This should pretty much reflect the develop branch at this point. Could you give it a read-over and let me know what should be changed?

damosvil, 2024-01-07:
It seems good to me

hyp0dermik-code, 2024-06-28:
Thoughts on changing the order to:
- User Notes (not sure if this is the right name for this section?)
- Bugfixes
- QoL
- Safety Features
- Misc
- Findings

Looking for feedback on the 'tuning notes' too - not sure how strongly this should be made. Based on my experience with introduced noise and high D Term noise I don't feel as if it needs to be particularly strong, merely a suggestion

Suggested changes as below:

User Notes:
- All pilots are suggested to check their tunes using their preferred method due to the improvements in zero crossing detection. If you are running a particularly aggressive tune (including some factory tunes and presets!), you should consider lowering your master multiplier for the first few flights.
- Startup power has been lowered to safe values for typical 4s-6s 5" ESCs and motors.
- If you experience problems with motors failing to start, read the [setup guide](https://github.com/bird-sanctuary/bluejay/wiki/Setup) to find suitable motor idle and startup power values for your quad.

Bugfixes: 
- ~~When fixing 3D mode, startup motor protection was inadvertently disabled which unfortunately lead to a couple fried ESCs and AIOs.~~ The bugfixes for startup motor protection appear to have an impact on 3d performance again - see **Known Issues**

Known issues:
- Based on initial testing, 3d mode appears to be having performance issues from the bugfixes for startup protection since 0.20
- As such use of 3d mode cannot be advised

stylesuxx, 2024-07-02:
Thank you @hyp0dermik-code - I updated the initial post according to your suggestion

hyp0dermik-code, 2024-07-15:
Can we add the following to the top

Huge thanks to all testers, who combined have flown over 1250 packs on 0.21RC alone

And corrections:
(two seperate lines, should not be crossed out)
-  When fixing 3D mode, startup motor protection was inadvertently disabled which unfortunately lead to a couple fried ESCs and AIOs in 0.20. 
-  The bugfixes for startup motor protection appear to have an impact on 3d performance again - see Known Issues

hyp0dermik-code, 2024-07-15:
I think adding information about the changes to ZCD to here makes the most sense https://github.com/bird-sanctuary/bluejay/wiki/Demag-events#what-changed-in-bluejay-020

I think the related section could be minimized a bit based on the chances of this actually being a problem (using Bl32 vs AM32 as an example).

So that section could become:

- All pilots are suggested to check their tunes using their preferred method due to the improvements in zero crossing detection. See [here](https://github.com/bird-sanctuary/bluejay/wiki/Demag-events#what-changed-in-bluejay-020) for more information on what the changes mean

stylesuxx, 2024-07-19:
Release is done - wohoo, closing this one.

closed 2024-07-19 by stylesuxx