bird-sanctuary/bluejay #163: Temperature Feedback Issue

ellmotion, 2023-08-18:
### Describe the issue

Hello,
I am unable to get the temperature feedback from my ESCs in Betaflight. The temperature remains at 0.
I have properly enabled DShot EDT.
ESC: SpeedyBee BLS 50A
FC: SpeedyBee F405 V3
[image]
[image]
[image]
[BTFL_cli_20230818_132422_SPEEDYBEEF405V3.txt](https://github.com/bird-sanctuary/bluejay/files/12378757/BTFL_cli_20230818_132422_SPEEDYBEEF405V3.txt)

### Bluejay version

0.20

### ESC variant

J_H_50

### PWM frequency

24

### DShot bitrate

300

### Bidirectional DShot

On

### FC firmware

Betaflight 4.4.2

damosvil, 2023-08-18:
Please, try Betaflight 4.4.0. In Betaflight 4.4.2 It seems that shot
commands doesn't work fine

ellmotion, 2023-08-18:
You are right, I have flash BF in 4.4.0 and I have the temperature feedback.
Thanks
[image]

closed 2023-08-18 by ellmotion