bird-sanctuary/bluejay #167: Turtle Mode and changing motor direction with betaflight configurator not working after 19.2

X-ForceFPV, 2023-08-23:
### Describe the issue

FC is BetaFPV 1S F411 AIO v2.0 ELRS

Reversing motor direction with betaflight configurator does not work as well

19.2 20.0 and 20.1 are also giving below 1% errors in the motor tab while spinning. 

everything works well with 19.0

### Bluejay version

0.20.1

### ESC variant

A_X_5

### PWM frequency

96

### DShot bitrate

300

### Bidirectional DShot

Off

### FC firmware

Betaflight 4.5 Master

### Motor size

0802

stylesuxx, 2023-08-24:
I am a bit confused - you state you have below 1% errors, but bi-directional DSHOT is not enabled? You should not be seeing a value then at all.

I know BF 4.5 had some issues with DSHOT commands in general, am surprised to hear that it works with 0.19.0. We will have to investigate this further.

X-ForceFPV, 2023-08-24:
Excuse me, i missed the box, it is enabled.

background:
i wanted the new erpm corrections because i use rpm filter.

BetaFPV delivered the fc with bluejay 0.17(beta), i guess because of bb51

hyp0dermik-code, 2023-08-24:
@stylesuxx Can you please clarify whether you mean 4.4.2+ or 4.5 specifically

On 4.5 PR I get <1% errors until the motors are enabled and spinning, and then get 0%. Can you confirm you have the same behaviour?

IIRC I had this same issue on 4.4.2 and 4.4.0 as well, I'll confirm later.

@X-ForceFPV By 19.0 working, you mean 0% errors with the "I accept" slider enabled but nothing else in the Motors tab, and you've observed for longer than a minute?

X-ForceFPV, 2023-08-24:
> @X-ForceFPV By 19.0 working, you mean 0% errors with the "I accept" slider enabled but nothing else in the Motors tab, and you've observed for longer than a minute?
> 
> On 4.5 I get <1% errors until the motors are enabled and spinning, and then get 0%. Can you confirm you have the same behaviour?
> 
> Can you please confirm what commit of 4.5z?

at first it looks like that, but giving it more time it starts oscillate between 0 and 0,7% errors over ~ 4 seconds.
above 19.0 i have one motor with constant errors about 0.8%

is there anything to read about dshot errors?

my version is 2 days old. 
4.5 (latest)master

stylesuxx, 2023-08-25:
> @stylesuxx Can you please clarify whether you mean 4.4.2+ or 4.5 specifically
4.4.2+ I have only looked briefly into 4.5 - a lot was not working at that time. Currently most of my testing quads are on 4.4.0 and 4.4.1

hyp0dermik-code, 2023-10-03:
> at first it looks like that, but giving it more time it starts oscillate between 0 and 0,7% errors over ~ 4 seconds. above 19.0 i have one motor with constant errors about 0.8%
>
> is there anything to [...]

Did you enable EDT safety arm when you switched to 19.2+? If so, can you disable it temporarily and tested again?

Prajwal-Koirala, 2023-12-27:
**Here's a summary of the issue and key points from the GitHub thread:**

**Issue:**

- Turtle Mode and motor direction reversal not functioning correctly in Bluejay firmware versions 19.2 and above.
- Issue observed on BetaFPV 1S F411 AIO v2.0 ELRS flight controller with Betaflight 4.5 Master.
- DSHOT errors reported in firmware versions 19.2 and above.

**Potential Solutions and Troubleshooting Steps:**

1. **Downgrade Bluejay Firmware:** Consider temporarily downgrading to Bluejay version 19.0, as it reportedly works without issues.
2. **Disable EDT Safety Arm:** If enabled, temporarily disable EDT Safety Arm and retest.
3. **Review DSHOT Configuration:** Ensure correct DSHOT settings, specifically:
    - Bidirectional DSHOT enabled.
    - Appropriate ESC variant selected.
4. **Investigate Betaflight 4.5 Issues:** Be aware of potential DSHOT-related issues with Betaflight 4.5. Consider using a different firmware version if possible.
5. **Provide Additional Information:** If using a custom build of Betaflight 4.5, share the commit number for further analysis.
6. **Gather Configurator Debug Log:** Obtain a configurator debug log for detailed troubleshooting.
7. **Seek Further Assistance:** If the issue persists, engage with Bluejay developers or seek help from the Betaflight community for more comprehensive troubleshooting.

**Additional Considerations:**

- **Acceptable DSHOT Errors:** Clarify whether any percentage of DSHOT errors is considered acceptable.
- **Investigation by Bluejay Developers:** Note that Bluejay developers are actively investigating the issue.

**I'll continue to monitor for updates and provide additional information as it becomes available.**