bird-sanctuary/bluejay #177: EDT safety arm prevents motors arming subsequently

hyp0dermik-code, 2023-10-04:
### Describe the issue

EDT safety arm prevents motors arming subsequently in turtle mode and motors tab in all versions tested down to 4.4.0

In 4.5 master the motors tab does not work on first arm

### Bluejay version

.20.1RC2

### ESC variant

A_X_5

### PWM frequency

96

### DShot bitrate

300

### Bidirectional DShot

On

### FC firmware

Betaflight 4.4.0

hyp0dermik-code, 2023-10-04:
I am not sure if this is intended behaviour or not, or whether it should be if it wasn't

damosvil, 2023-10-04:
I have reproduced the issue on my side. I will work to fix it.

stylesuxx, 2023-10-04:
Should we backport this to v0.20.1?

damosvil, 2023-10-04:
Yes. I tried with v0.20.1. Tomorrow I will start debugging it.

damosvil, 2023-10-07:
These are the assets for the fix in:
https://github.com/bird-sanctuary/bluejay/pull/178

Please, could you check on your side?
[Bluejay_0.20.1_RC3.zip](https://github.com/bird-sanctuary/bluejay/files/12837165/Bluejay_0.20.1_RC3.zip)

closed 2023-11-16 by stylesuxx