bird-sanctuary/bluejay #188: EDT - voltage and current not being received in Ardupilot

GalacticGreg, 2024-03-13:
### Describe the issue

Hi,
I'm using 0.21RC1 and arducopter 4.5beta2.

Ardupilot isn't receiving current or voltage but it is receiving temperature and RPM through BDshot EDT and EDT handshake is working.

Before I submit a bug in ardupilot, I thought I'd check to see that it works in Betaflight or inav and to see if there are other factors that may affect this.

Thanks!

### Bluejay version

0.21RC1

### ESC variant

A_H_30

### PWM frequency

24

### DShot bitrate

600

### Bidirectional DShot

On

### FC firmware

Arducopter 4.5beta2

### Motor size

2205

stylesuxx, 2024-03-13:
There is currently no ESC that would support this functionality with Bluejay.

closed 2024-03-13 by stylesuxx

GalacticGreg, 2024-03-13:
Ok, thanks!