bird-sanctuary/bluejay #191: LED control via DSHOT

stylesuxx, 2024-04-08:
LED control is currently only possible through config, should also be possible via DShot commands.

edermi, 2024-04-19:
How can I enable / configure this currently with BlueJay? I have a T1 Ranger and the used ESCs apparently have a LED I'd like to turn on.

stylesuxx, 2024-04-19:
On specific layouts you can set the LEDs via configurator, yes. If you link the ESC, I should be able to tell you with more confidence...

edermi, 2024-04-19:
Should be this https://www.heewing.com/products/hee-wing-rc-t1-ranger-vtol-fx-25a-upgraded-2022-2023-esc

According to https://youtu.be/e_ng6lvXPlw there's an LED, he's using the BLHeli configurator to turn it on

stylesuxx, 2024-04-20:
Unfortunately it does not say which layout this is, but if it works in blheli_S it should work on Bluejay too.

edermi, 2024-04-20:
That's my hope 🙂
Can I use the same software for configuration?

stylesuxx, 2024-04-20:
You'd use https://esc-configurator.com

stylesuxx, 2025-10-02:
If it's the correct layout, the option should be there. Which layout does it show?

Meta-Cake, 2025-10-02:
It is showing  J-H-30 whenever I read the settings back

Meta-Cake, 2025-10-02:
It showed up! Thank you, I didn't even realize there was an individual option in the configurator. It might help some other people if there were a little note where the option should be that said, "This ESC supports LEDs; to configure them, select 'Disable common settings' in the settings menu," or something similar