bird-sanctuary/bluejay #192: ESC sometimes will not respond to bi-directional DShot signal.

wjxway, 2024-04-14:
### Describe the issue

I am trying to read the eRPM feedback from bluejay esc using bi-directional dshot but it appears that sometimes the esc will not respond to my commands (the motor is spinning fine though, but I am also not changing my command all the time.). I am not sure if ESC never received my signal correctly when it's not responding or if it sometimes will choose not to respond? Is this a bug?

The symptom can be best seen in the following figures. The first figure shows the comparison between two consecutive and identical transmissions, the ESC choose to respond to the first but not the second. The waveforms for these two transmissions can be seen in the second and third image and I don't think there is a difference. Sometimes it will respond to all messages in a period and then not respond for a while, and sometimes it suddenly respond to one transmission but not the other.

[image]
[image]
[image]

Also I am curious about the details of DShot protocol. Specifically what is the acceptable range of pulse width and period, and how ESCs detect and decipher the signal? Are there any time sequence plots for the protocol similar to the ones in chip datasheets? I expect that if you change the pulse width a little bit it shouldn't be a problem for the ESC as it would still be clear what is 0 and what is 1, but apparently that is not the case and I frequently encounter situations where the ESC simply refuse to accept the signal just because the pulse width is off by several tens of ns.

Thanks a lot devs!

### Bluejay version

0.19.2

### ESC variant

P_H_10

### PWM frequency

96

### DShot bitrate

Any

### Bidirectional DShot

On

### FC firmware

Custom

### Motor size

T-Motor 1103 KV8000

stylesuxx, 2024-04-15:
This sounds to me like a timing issue to be honest. The timings need to be kept very tight, otherwise things are out of whack and such a result would be expected.

Please have a look at the [DSHOT documentation](https://betaflight.com/docs/development/api/dshot). You can also have a look at DSHOT implementation withing Betaflight.

If you need further help, please post a link to your code.

wjxway, 2024-04-16:
I see, it seems that the length of the whole signal is off by ~1us and that might be the cause. Let me revise that and see if it can work better.

I think another cause is that the ESC will not respond with eRPM data if it has just started up, I have to turn it on and let the motor spin a little bit to make eRPM appear constantly.

I still want to complain about the documentation of DShot. It is so unclear that I have to guess all the time what the designer is doing, especially in the eRPM part. Some sources say that the eRPM's CRC is `(data^(data>>4)^(data>>8)&0xFF`, but it is actually `(~(data^(data>>4)^(data>>8))&0xFF`.

I will try to write a more detailed explanation with examples of DShot protocol when I have time.

wjxway, 2024-04-16:
I think I figured it out. I used a pattern where I accelerate and decelerate the motor every other command so if there's a lost packet I will be able to hear it. I can confirm that the ESC never received my transmission so it never responded.

Although I tried various pulse width that should be very close or even identical to the documentations, it still fails about 20% of the time, then I tried to switch to DShot 300, and it immediately worked flawlessly.

It seems that DShot 600 is not nearly as stable as DShot300, when I switched to DShot300 almost all problems are solved. Now the only time when it will fail is when I suddenly and drastic accelerate the motors, and it is probably caused by several flywires on my test board.

Fortunately I don't really need to change my command every 50us, so DShot 300 would work perfectly for me. Thanks for your help @stylesuxx !

closed 2024-04-16 by wjxway