bird-sanctuary/bluejay #193: Request restart BLHeli_S update

tmotorsg, 2024-06-05:
Hello everyone, I am from Asia, I am a distributor in our local community, specializing in providing hardware for FPV.

This is my first time on GitHub. This post is Request restart BLHeli_S update.

We have a large number of users who are still using 8-bit ESCs. We are also the distributor of HAKRC. HAKRC comes with BLHeli_S firmware when it leaves the factory. BLHeli_S has stopped updating for many years.

We have done a lot of tests on Bluejay and BLHeli_S. We tested 50 batteries with the same hardware and the same battery. We found that BLHeli_S is more stable than Bluejay, but it does not have as many functions as Bluejay. The firmware of BLHeli_S is stronger in the pulling force part, and the motor is more violent，We have been notified of the BLHeli_32 issue, but this does not affect our 32-bit ESC users, because we also tested it on AM32, but we will not discuss the 32-bit issue now.

If anyone in the community is willing to start updating BLHeli_S and perform maintenance, we as a distributor can provide hardware sponsorship!

Add some functions based on the existing BLHeli_S

Add PWM frequency（48 96kHz PWM）

Increase DShot bitrate

We can provide 8-bit ESC hardware sponsorship，Now a large number of brands still ship with BLHeli_S firmware

We do have a large number of users who continue to use BLHeli_S, which is an open source project

Richard

stylesuxx, 2024-06-05:
This is the Bluejay repository. If you want to talk about Bluejay we can do this.
Your statements make little sense:
- what does "more stable" mean?
- where are the logs?

I am closing this issue. If you have concrete feedback, please open a proper issue with all required information. Also feel free to join us on Discord.

closed 2024-06-05 by stylesuxx

tmotorsg, 2024-06-06:
> This is the Bluejay repository. If you want to talk about Bluejay we can do this. Your statements make little sense:
>
> * what does "more stable" mean?
> * where are the logs?
>
> I am closing this issue. [...]

Sorry, I posted it in the wrong place

We do have 16G test data, including videos to prove it, and we have a power motor test bench

But our target is not Bluejay, but BLHeli_S

tmotorsg, 2024-06-06:
Sorry, I posted it in the wrong place
We do have 16G test data, including videos to prove it, and we have a power motor test bench
But our target is not Bluejay, but BLHeli_S
---- Replied Message ----
| From | Chris ***@***.***> |
| Date | 6/6/2024 00:41 |
| To | ***@***.***> |
| Cc | ***@***.***>,
***@***.***> |
| Subject | Re: [bird-sanctuary/bluejay] Request restart BLHeli_S update (Issue #193) |

This is the Bluejay repository. If you want to talk about Bluejay we can do this.
Your statements make little sense:

what does "more stable" mean?
where are the logs?

I am closing this issue. If you have concrete feedback, please open a proper issue with all required information. Also feel free to join us on Discord.