bird-sanctuary/bluejay #194: One question about the bluejay firmware

fy1782281848, 2024-06-13:
what is the difference between the J-H-50-24/48/96 three, and whether they can be directly replaced by the J-H-50 BLHeli_S revision 16.7

stylesuxx, 2024-06-13:
Yes, you can switch between Bluejay and BLHeli_S as long as you flash the same **Layout-MCU-Deadtime** combination. Just be aware to adjust your Flight-Controller accordingly when flashing BLHeli_S (disable RPM filtering...).

closed 2024-06-13 by stylesuxx

fy1782281848, 2024-06-14:
Thank you sir, I have another question, can BB21 be replaced by BB51? They can use the same firmware without modifying the hardware design

stylesuxx, 2024-06-14:
No, unfortunately BB51 is not a drop in replacement. There are specific layouts for BB51.

howels, 2024-06-15:
> No, unfortunately BB51 is not a drop in replacement. There are specific layouts for BB51.

But BB51 is supported by BlueJay, using the X series of layouts. So it is supported, but requires a different firmware target.

fy1782281848, 2024-07-31:
Hello, sir. Could you please explain why the J-H-50 firmware for single ESC
does not operate properly when using PWM directly? What settings need to be
adjusted?

stylesuxx, 2024-07-31:
Bluejay only supports DSHOT protocol.

fy1782281848, 2024-09-09:
Can the P0.5 pin of BB21 be used as the throttle signal input?