bird-sanctuary/bluejay #203: Docker - Licensing issues with non matching OS versions

hyp0dermik-code, 2024-07-24:
With Keil licensed in Debian 12, and Dockerfile set to Debian:10:
```
./build.sh -l J -m H -d 40 -p 48 
Building J_H_40_48
d5ea59b4ea8ceb8abb604b6fc6b4ec06b8ab64a1bb13bb55aa233c0882763642
fatal: detected dubious ownership in repository at '/root/source'
To add an exception for this directory, call:

	git config --global --add safe.directory /root/source
AX51 : build/J_H_40_48_.OBJ
LX51 : linking build/J_H_40_48_.OBJ to build/J_H_40_48_.OMF
*** WARNING L30: MEMORY SPACE OVERLAP
    SEG:     ?DT?BLUEJAY?21
    FROM:    I:000020H   
    TO:      I:000028H   
--
*** WARNING L30: MEMORY SPACE OVERLAP
    SEG:     ?DT?BLUEJAY?21
    FROM:    I:000020H   
    TO:      I:000028H   
--
*** ERROR L250: CODE SIZE LIMIT IN RESTRICTED VERSION EXCEEDED
    LIMIT:   0800H BYTES
make: *** [Makefile:132: build/J_H_40_48_.OMF] Error 1
```

With Keil licensed in Debian 12, and Dockerfile set to Debian 12:
```
./build.sh -l J -m H -d 40 -p 48 -c user/Settings
Building J_H_40_48
ec9526df5ba9e2b52ec938d9fd05ba501367b1689a0fdb266ac5bd5fd994ac6b
fatal: detected dubious ownership in repository at '/root/source'
To add an exception for this directory, call:

	git config --global --add safe.directory /root/source
AX51 : build/J_H_40_48_.OBJ
LX51 : linking build/J_H_40_48_.OBJ to build/J_H_40_48_.OMF
OHX  : generating hex file build/hex/J_H_40_48_.hex
rm build/J_H_40_48_.OMF
```

hyp0dermik-code, 2024-07-24:
Any fix will likely break the current Github build environment and require the regeneration of the .wine folder

stylesuxx, 2024-07-24:
I am a bit surprised by this to be honest. Why would this make a difference? Is the structure of the .wine dir different?