bird-sanctuary/bluejay #205: Complementary PWM bug

guser210, 2024-08-07:
### Describe the issue

We are testing motors for long range flying and came across an issue that happens often at 24KHz PWM frequency, we've noticed an invalid PWM output from the MCU where complementary PWM stops working and outputs high on both lines for a full PWM pulse, it happens regularly on 24KHz and is not present on 48KHz. the anomaly happens on all 3 pwm outputs.

PicoScope files are over 100MBs in size and unable to upload.
Complementary PWM on all drive lines, last line is current dips when the anomaly is present
[image]
[image]

Same log file Zoomed out, the spikes on the bottom are the anomalies in the lines.
[image]

48KHz without PWM issues.
[image]
[image]

### Bluejay version

0.18

### ESC variant

I_L_15

### PWM frequency

24

### DShot bitrate

300

### Bidirectional DShot

On

### FC firmware

Betaflight 4.4.0

### Motor size

3210

guser210, 2024-08-07:
Complementary PWM outputs from MCU.
[image]

ot0tot, 2024-08-07:
https://github.com/bird-sanctuary/bluejay?tab=readme-ov-file#deprecated-platforms

guser210, 2024-08-07:
Unfortunate, 
Thank you.

stylesuxx, 2024-08-08:
Might be related to #187 
As stated by @ot0tot L MCUs are deprecated.

Any chance you could replicate this on H/X MCUs with the current Bluejay version?

guser210, 2024-08-08:
> Might be related to #187 As stated by @ot0tot L MCUs are deprecated.
> 
> Any chance you could replicate this on H/X MCUs with the current Bluejay version?

Replicate the test data or the condition? I do not have any ESC with H/X MCUs but I do have a whole lot of ESCs with L MCUs that need to find a solution for.

ot0tot, 2024-08-09:
Can you not use 48kHz PWM?

If this is for long range flying, why not use newer, more reliable hardware?