bird-sanctuary/bluejay #206: Failed Fetching Firmware Y-H-10 all versions

wafflemcbutterpants, 2024-08-08:
### Describe the issue

Trying to flash any available version of Y-H-10 results in "Could not get file for flashing" in the configurator and log reports "Failed Fetching Firmware."
Manually searching for the .hex doesn't result in any Y-H-* files being found. 

### Bluejay version

21.0

### ESC variant

Y-H-10

### PWM frequency

Any

### DShot bitrate

Any

### Bidirectional DShot

Any

### FC firmware

BTFL version: 4.4.0

### Configurator debug log

2024/08/07 22:34:43: Chrome (127.0.0.0) on Windows
2024/08/07 22:34:44: Plugged in
2024/08/07 22:34:46: Opened serial port
2024/08/07 22:34:46: MultiWii API version received: 1.45.0
2024/08/07 22:34:46: Flight controller info, identifier: BTFL version: 4.4.0
2024/08/07 22:34:46: Running firmware released on: Apr 19 2023 14:58:34
2024/08/07 22:34:46: Board: BeeB, version: 0
2024/08/07 22:34:46: Unique device ID received - 0x04b004a3139511735343435
2024/08/07 22:34:51: Trying to read 4 ESCs
2024/08/07 22:34:52: Read ESC 1: Y-H-10 - BLHeli_S, 16.7
2024/08/07 22:34:53: Read ESC 2: Y-H-10 - BLHeli_S, 16.7
2024/08/07 22:34:54: Read ESC 3: Y-H-10 - BLHeli_S, 16.7
2024/08/07 22:34:55: Read ESC 4: Y-H-10 - BLHeli_S, 16.7
2024/08/07 22:34:55: Done reading ESCs
2024/08/07 22:35:10: Chosen firmware: https://github.com/bird-sanctuary/bluejay/releases/download/v0.21.0/Y_H_10_96_v0.21.0.hex 
2024/08/07 22:35:11: Failed fetching firmware 
2024/08/07 22:35:11: Could not get file for flashing.

stylesuxx, 2024-08-08:
Y layout does not exist.
What ESC is this? Please link to product page.

wafflemcbutterpants, 2024-08-08:
It’s the original NewBeeDrone AcroBee65 BeeBrainBL Brushless (Now called BLV1 but wasn’t at the time). NBD doesn’t have a product page anymore but this is the same product as here:
https://www.getfpv.com/newbeedrone-acrobee65-beebrainbl-brushless-micro-drone-bnf.html

stylesuxx, 2024-08-08:
Yeah, I can't help you out here unfortunately. You could ask them to reach out to us and commit the Y Layout. We would also need a sample to verify functionality.

Without the layout information we are unable to build a firmware.

And you are sure, that the new version of this computer also uses the Y layout?

stylesuxx, 2024-11-06:
Closing this issue.

Y layout not supported. If NBD decides to share it we can add it.

closed 2024-11-06 by stylesuxx