bird-sanctuary/bluejay #217: Unsupported Firmware (Layout 206) is not yet supported

rpisarew, 2024-09-17:
### Describe the issue

I have two new [Air 5in1](https://betafpv.com/collections/components/products/air-brushless-flight-controller?variant=41142912745606) Boards. One of them works without any issues, but with the second I have a issue.

I have one esc where the configurator is not able to recognize the firmware version:
[image]

The esc is working in flights until now without any issues, I am just not able to update the firmware..
Can I safely do something like a force flash to fix that, without bricking the esc?

### Bluejay version

0.19

### ESC variant

A-X-5

### PWM frequency

96

### DShot bitrate

300

### Bidirectional DShot

On

### FC firmware

Betaflight 4.5.1

### Configurator debug log

1155:22336

115200
Open Port Selection

Disconnect

English
Settings
Hide Log
2024-09-17 @ 20:50:49 -- Done reading ESCs
2024-09-17 @ 20:50:49 -- Read ESC 4: A-X-5 - Bluejay, 0.19, 96kHz
2024-09-17 @ 20:50:49 -- Read ESC 3: A-X-5 - Bluejay, 0.19, 96kHz
2024-09-17 @ 20:50:49 -- Read ESC 2: A-X-5 - Bluejay, 0.19, 96kHz
2024-09-17 @ 20:50:49 -- Read ESC 1: UNKNOWN
2024-09-17 @ 20:50:49 -- Layout revision 206 is not yet supported
2024-09-17 @ 20:50:48 -- Trying to read 4 ESCs
2024-09-17 @ 20:50:24 -- Unique device ID received - 0x024004d4232500120303059
2024-09-17 @ 20:50:24 -- Board: SG47, version: 0
2024-09-17 @ 20:50:24 -- Running firmware released on: Aug 4 2024 19:03:57
2024-09-17 @ 20:50:24 -- Flight controller info, identifier: BTFL version: 4.5.1
2024-09-17 @ 20:50:24 -- MultiWii API version received: 1.46.0
2024-09-17 @ 20:50:24 -- Opened serial port
2024-09-17 @ 20:50:20 -- Plugged in
2024-09-17 @ 20:50:09 -- Chrome (128.0.0.0) on Mac OS

stylesuxx, 2024-09-17:
You got it like this out of the box? I would RMA it. Technically you should be able to just flash A-X-5 to the on ESC and it should be fine.

rpisarew, 2024-09-17:
> You got it like this out of the box? I would RMA it. Technically you should be able to just flash A-X-5 to the on ESC and it should be fine.

Yes it was like that out of the box. I just tried the force flash and it fortunately worked :)

closed 2024-09-17 by rpisarew