bird-sanctuary/bluejay #218: "Bluejay (.0) version '0.21' (Layout 208) is not yet supported.

winst0niuss, 2024-10-31:
### Describe the issue

Just flashed my Air75 to bluejay 21 and now i get this error message: 
"Bluejay (.0) version '0.21' (Layout 208) is not yet supported.

### Bluejay version

0.21.0

### ESC variant

A-X-5

### PWM frequency

96

### DShot bitrate

300

### Bidirectional DShot

On

### FC firmware

Betaflight 4.5.0

### Motor size

0802

### Configurator debug log

[esc-configurator-log.txt](https://github.com/user-attachments/files/17585054/esc-configurator-log.txt)
[image]

stylesuxx, 2024-10-31:
Comfigurator version seems wired. Which URL are you flashing from?

winst0niuss, 2024-10-31:
https://esc-configurator.com/

stylesuxx, 2024-10-31:
Then please hard refresh the page. Bottom right should show version 0.32.1

closed 2024-10-31 by stylesuxx