bird-sanctuary/bluejay #219: EDT arming stop motor all the time

PascalRZ, 2024-11-01:
### Describe the issue

Hello,
Because 2 weeks ago I was seriously injured at my left hand by one of my quad with one or two motors starting for 1 second at high speed with BLHeli_S 16.6 + Betaflight 4.2.11 quad disarmed, I’m updating all my quads with BLHeli_S to BF 4.5.1 and BlueJay 0.21.0 (and quad with BLHeli_32 to AM32). I hope “EDT safety arm” will improve quad safety when disarmed in BF. BlueJay team members, thank you for your job!
But, I installed BlueJay 0.21.0 on two quads (one Eachine X140HV with G-H-30 and one Eachine X220S with A-H-25). When “EDT safety arm” is checked:
-	On the X140HV, if motors don’t succeed to start at the first try of BlueJay “start power protection”, they only make noise and move a little backward-forward. On the X140HV, with the two start power protection values at the max (yes, the max!), motors never start, they only make noise and move a little backward-forward.
-	If motors start on the X220S, and if I use motor_stop in BF4.5.1, when I reduce to min throttle, motors stop but when I put throttle again, they only make noise and move a little backward-forward. I have to disarm and arm in BF to see the motors starting again. It means that when I will be in flight, if I reduce throttle too much, motors will stop and will not restart, and my quad will fall from the sky and crash!!! I don’t want to try in flight, I did my tests on the ground.

I tried with BF dshot_edt=on and off without difference.
When EDT safety arm is unchecked, everything works well, my quads fly :)
Note: I can see bi-dir DShot RPM and ESC CPU temp values in BF motor conf, when I allow motors to spin and are spinning.
A friend has the same issue with another quad (I don’t know his configuration).
Hope you will find the bug and correct it, maybe with BF team, because “EDT safety arm” is a major update in safety.
Pascal 

### Bluejay version

0.21.0

### ESC variant

G-H-30 and A-H-25

### PWM frequency

48

### DShot bitrate

300

### Bidirectional DShot

On

### FC firmware

Betaflight 4.5.1

### Motor size

1507-2500KV and MN2206-2300KV

stylesuxx, 2024-11-01:
Hey @PascalRZ sorry to hear you got injured.

As @ot0tot said, make sure to follow the setup guide regarding startup power. I'd disable safety arm until you set startup power correctly, then you can exclude it as error source. Enable it once you have the startup power settings dialed in.

PascalRZ, 2024-11-01:
Hi ot0tot !
Please find Submit_support_data_c2155526-c333-497a-92a0-a1bd6c282194.txt and X140HV_BlueJay0210_printscreen.docx for a working conf of my X140HV. If I check "Forcer l'armement EDT" and "Ecrire les paramètres" and I restart my quad (battery removed and reconnected) motors will not start at all (start power protection value from min to max). No difference with BF dshot_edt=on or off.

[Submit_support_data_c2155526-c333-497a-92a0-a1bd6c282194.txt](https://github.com/user-attachments/files/17604084/Submit_support_data_c2155526-c333-497a-92a0-a1bd6c282194.txt)

[X140HV_BlueJay0210_printscreen.docx](https://github.com/user-attachments/files/17604088/X140HV_BlueJay0210_printscreen.docx)

ot0tot, 2024-11-01:
Your CPU usage is concerningly high, likely due to poor FC design preventing the OSD from using DMA. I would consider reducing your PID loop rate to 2kHz, and or overclocking to 120MHz (`set cpu_overclock = 120MHZ` and `save` in the CLI). 

The inconsistent starting behavior is likely due to your DShot idle value being set to low. As recommended in the Bluejay setup guide, this should be increased to 5.5%.

PascalRZ, 2024-11-01:
@ot0tot

Your CPU usage is concerningly high, likely due to poor FC design preventing the OSD from using DMA. I would consider reducing your PID loop rate to 2kHz, and or overclocking to 120MHz (set cpu_overclock = 120MHZ and save in the CLI).
--> Yes it's true, I have noticed that, I just updated the PID loop to 2 kHz, CPU charge is 31-32% now.

The inconsistent starting behavior is likely due to your DShot idle value being set to low. As recommended in the Bluejay setup guide, this should be increased to 5.5%. 
-->This value was set to 5.5% in BF (default parameter) when I had the problem. My setting at 4% works very well without "EDT safety arm". Note : 1000*(1+4/100)=1040, not exotic.

The main difference between my two quad's motors is X140HV 1507-2500KV motors have an important "magnetic resistance" unpowered, X220S MN2206-2300KV don't have.

PascalRZ, 2024-11-01:
Another information : the problem happens with BF and when I use ESC-configurator to control motors (with Chrome navigator).

stylesuxx, 2024-11-01:
> Another information : the problem happens with BF and when I use ESC-configurator to control motors (with Chrome navigator).

That's not how you should be setting things up though. You should test and setup during regular arm process. Motor control in the configurator just sends commands via BF but does not involve the pid loop. Motor control there is just for quickly checking direction.

PascalRZ, 2024-11-01:
@stylesuxx 
Ok but the real problem is not if motors start spinning or not, the problem is that without "EDT safety arm" everything works well, but as soon as I check "EDT safety arm" : If motors start on the X220S, and if I use motor_stop in BF4.5.1, when I reduce to min throttle, motors stop but when I put throttle again, they only make noise and move a little backward-forward. I have to disarm and arm in BF to see the motors starting again. It means that when I will be in flight, if I reduce throttle too much, motors will stop and will not restart, and my quad will fall from the sky and crash!!! I don’t want to try in flight, I did my tests on the ground.

I'm not a programmer so I can't read the code but it looks like:
- BF send the handshake when the quad is armed, but only on time
- each time a motor stop spinning (voluntary when I put min throttle with motor_stop enabled, or not voluntary on a desynch), BlueJay is waiting for a new arm handshake but BF doesn't send it because it is already armed. BlueJay, without a new arm handshake, doesn't accept to restart the motor and makes a beacon sound
- If I disarm, BF send a disarm handshake
- If I arm, BF send an arm handshake
- BlueJay accepts to spin the motors.
- etc...

stylesuxx, 2024-11-01:
Yeah, don't use motor stop, things should be working as expected then.

PascalRZ, 2024-11-11:
Hi @stylesuxx @ot0tot !

I did new tests:
I confirm you must “set dshot_edt=on” and “save” in BF CLI when you want to use “EDT safety arm” in BlueJay. If you use the default parameter dshot_edt=off, motor never spin and make a beacon noise (no edt = no arm handshake = it’s normal).
When I arm/disarm in Air Mode (so without BF Motor Stop), motors start each time I arm and restart automatically when I stop one of them with my hand (and without props of course ;) ) to simulate a gate collision. -> No problem
When I use Motor Stop, or with BetaFlight Configurator motor command, or with ESC Configurator motor command, motors start only one time from 0 to what you want, and when you return to 0 and want to restart, motors do a beacon sound and don’t spin.
I don’t know when the arm and disarm handshake is sent but something goes wrong, not necessary in BlueJay but maybe yes, or the bug is in BF/BF Conf/ESC Conf.

What I can’t simulate is a 0 order of BF during a flight when using dynamic idle (I don’t use it and I think it’s hard to reproduce)
https://betaflight.com/docs/wiki/guides/current/dynamic-idle
“With dynamic idle enabled, the PIDs are allowed to send zero drive to the motors”. It means the ESC will receive a 0 order, same value as Motor_Stop, and will not spin again, quad falling from the sky and crashing… Is it true?

Do you think the problem is in BlueJay or in BF/BF Conf/ESC Conf?

And of course, pilots can use BF Motor_Stop if they want ;)

stylesuxx, 2024-11-11:
Safety arm check are here: https://github.com/bird-sanctuary/bluejay/blob/1b0802dbbf28f27572ba08b0a292db8adae29652/src/Bluejay.asm#L687 

Bluejay will not arm, while `Flag_Ext_Tele` is not set (when EDT safety arm is enabled).

So every-time this flag is cleared, arming will not work until it is set again. This flag is cleared under the following conditions:

* Initial setup phase
* EDT disable command is received
* exiting run mode after stall
* ` Flag_Rcp_Stop` is set - this can happen for the following reasons: pulses are not recognized as DSHOT, error during startup, dshot signal timeout

So when triggering from configurator, the initial handshake works fine, you can spin motors. You return to 0 and then at some point most likely timeout is triggered.

> It means the ESC will receive a 0 order, same value as Motor_Stop, and will not spin again, quad falling from the sky and crashing… Is it true?

No this is not true. While in operation, the ESC will receive continuous DSHOT signal, even if the value is 0. This does NOT reset the  `Flag_Rcp_Stop` flag.

Are you seeing this issue issue in real life? Or are you just having the issue in the configurator? If just in the configurators, this is kind of expected.

PascalRZ, 2024-11-11:
@stylesuxx 

I race in acro mode using motor_stop and without air mode because I only reduce throttle stick to 0 when I crash and I stop props faster than moving my hand to the arm/disarm switch. It’s a long discussion between pro and cons but many pilots do that. I use air mode only when I do freestyle because I have to reduce throttle to 0 keeping quad attitude under control when I do some tricks and because I’m not at 50cm of the ground when I do it.
Of course, sometime, when I race, I reduce my throttle stick too low and motors stop spinning. As soon as I push the stick, motors spin again and I have control in less than one second. When “EDT safety arm” is selected in BlueJay, motors make beacon sound instead of spinning again and I my quad crashes instead of flying. Yes, it’s real life!!!

For BF Configurator and ESC Configurator, it could be a good idea to warn the user that motors will only spin one time and then will do a beacon sound to confirm EDT safety arm works well (and some user will not think BlueJay is bugged).

I understand that BF stops sending continuous DSHOT signal when throttle is at zero with motor_stop enabled despite quad is armed. Is it true? I don’t know how to test it. Do you think I have to open an issue in BF?

stylesuxx, 2024-11-11:
>  When “EDT safety arm” is selected in BlueJay, motors make beacon sound instead of spinning again and I my quad crashes instead of flying. Yes, it’s real life!!!

Mhh, that's interesting - could you post some logs please.

@damosvil could I ask you to jump in here? Are we resetting the `Flag_Rcp_Stop` flag because we are exiting the run loop on 0 throttle?

> Do you think I have to open an issue in BF?
That might indeed need to be the case. I think the initial handshake works fine and we would then need to re-initiate the handshake after coming back from 0